bwack/C64-Saver-bwack #4: Firmware for the addon board

bwack, 2018-07-23:
Programming the ATTiny85

We will use the Ardino IDE or avr-gcc toolchain.
For OLED display and ATTiny we will base our code on this project: 
http://cyaninfinite.com/tutorials/interfacing-a-i2c-0-96-oled-display-with-attiny85/ or https://www.instructables.com/id/ATTiny85-connects-to-I2C-OLED-display-Great-Things/

9VAC current and 5V current and voltage is read over I2C.
9VAC voltage is sampled on ADC. On the same ADC line the use button is attached.

discoHR, 2018-09-03:
In your latest YouTube video you mentioned menu navigation. A single button is not enough for that. Can we have a pushable rotary encoder instead of the button? Are there any pins left on AtTiny?

discoHR, 2018-09-03:
Actually we could use a single button (long press to enter selection mode, single press in selection mode to change option and long press again to select it) but that would be awkward, especially for selecting cutoff voltage.

bwack, 2018-09-03:
I am very happy that you like to help on this. Very much appreciated !

discoHR, 2018-09-03:
No problem. Please send me the boards so I can start. Let me know (on Facebook or in mail) if there are any additional components I'll need. I'm going to order some unrelated stuff from Mouser in the next couple of days so I could order these components as well.

bwack, 2018-09-03:
Sounds great. Can you start with a breadboard setup ? I can send you display, ina219 breakout board and an attiny85 soldered on dip8 adapter. Do you have a usbasp or similar ?

discoHR, 2018-09-03:
I have only TL866A. I guess I could do it on a breadboard but I'd rather do it on a proper board, even if it means waiting longer.

bwack, 2018-09-03:
OK. 

I noticed yesterday that I could not progam the attiny with ICSP. The setting for using ICSP was disabled/grayed out. Was able to do it with atmega328p, but not attiny85.

discoHR, 2018-09-03:
U2 seems to be interfering with ICSP. See linked thread for details.
https://www.eevblog.com/forum/projects/issue-programming-attiny-using-icsp/

hansdals, 2018-09-05:
That issue is different. The guy on the eevblog forum had a driver on the sck line, while we have a I2C isolator which is both high impedance and driving. The line should be high impedance normally. Lets see if it works.

hansdals, 2018-09-06:
If you like can you see if you can add the TinyWireW library to the project? I't is a C++ wrapper for another TWI library (coded in C) I think comes with the avr-gcc installation. Or we could use that library directly. Or go C++..

bwack, 2018-09-06:
Another USI-I2C code and interesting read about I2C
https://www.instructables.com/id/ATTiny-USI-I2C-The-detailed-in-depth-and-infor/
https://github.com/CalcProgrammer1/Stepper-Motor-Controller/tree/master/UnipolarStepperDriver

discoHR, 2018-09-07:
Thanks. This is a great I2C guide. Which slave I2C addresses are we using and which values can I expect from them?

Regarding libraries and mixing C with C++, I'm not a fan of either. I2C seems to be simple enough, it's just another way of bitbanging. I'm sure I can write my own code for that. Same for driving the OLED, I've done it in VHDL so doing it again in C should be straightforward. Using libraries is ok for complex stuff.

bwack, 2018-09-07:
if you use the usi hardware module, I think you only need to bit bang the serial clock. a bit weird how that works and that it is not fully hardware controlled.

closed 2018-11-30 by bwack