bwack/C64-Saver-bwack #15: Code locks up when using the sd1306 and ina216 libraries together

bwack, 2018-10-20:
@discoHR From what I remember when debugging this I saw that after the sd1306 library had the i2c and the ina216 was taking over, the code got stuck in a wait for high or wait for low call.. The libraries works fine on the. Find out why one module screws up for the other.

Btw this is why I have included the usi_i2c_master library. This library is from the nunchuk64 project.

bwack, 2018-10-20:
I think it got stuck here. sorry too long ago I don't remember.. I suspect the ssd1306 libray, or the way we use it. maybe it needs to be freed or something. At least the i2c_master expects a state that it never gets. Could be a state set in the TWI register. Not sure.

i2c_master infinite loop problem:
```
	while (!(PIN_USI & (1<<PIN_USI_SCL)));		//Wait for SCL to go high

	#ifdef I2C_FAST_MODE
		USI_I2C_WAIT_HIGH();
	#else
USI_I2C_WAIT_LOW();
```

discoHR, 2018-10-21:
It's not the code, it's INA. I removed both usi_i2c_master and tinusaur-ssd1306xle code, disabled all interrupts and wrote my own I2C bitbanging code including autodetecting INA's slave address. There is aboslutely no response from INA at any slave address, it's not sending ACK at all. Thought it was working earlier but I was mistaking SDA going up for an ACK (technically, it's a NAK and it was going up because of the pullup, INA had nothing to do with it).

There is a lot of garbage on I2C bus immediately after power-up. Could be it's messing up INA's I2C but I added procedure for resetting INA's I2C bus before trying to talk to INA, same problem. INA's bus is reset when SDA or SCL are pulled low for at least 28ms.

Replaced R4 with 0 ohm resistor, same problem.

Checked optoisolator, signals seem to pass to both ends. There is about 4.2V on one end and 5V on the other but that shouldn't matter.

Removed optoisolator and tried accessing INA directly, same problem.

Either INA you sent me is dead or it's I2C doesn't work without a load. There is only 2.45 mV on the shunt resistor while the red LED is on, if it matters.

Ordered 2x INA219 from an EU seller, it will be faster than ordering from China.

Some screenshots...

I2C bus garbage on boot:
[image]

Autodetection of INA's slave address, no ACK was ever received:
[image]

Same as above but zoomed in. ACK missing. Pullup resistor is doing NAK, INA must pull it low for ACK.
[image]

bwack, 2018-10-22:
It could be both the ina and the issue I descibed above. Just beware of that.

bwack, 2018-10-22:
maybe the voltage on the ina's VDD is too low because of the drop-out voltage in the 5V regulator driven at V_IN=5V.

bwack, 2018-10-22:
Great work btw. Tell you what. I will hook up the addon-board to my c64 DIY PSU with ina219 voltage meter. I can easily unscrew the cables going to the ina and connect the addonboard and see if I get read-out on the PSU display (if you know, I have a C64 DIY PSU with ina219 and arduino pro micro)

bwack, 2018-11-02:
Taken from my C64 DIY PSU with only INA219 on the I2C bus. Data rate is 100kHz. Using adafruits ina219 library. You can see how it at very first writes calibration bits, and it does that several times too. Second transfer is the configuration h00. No reset command (that would be most significant bit set to h00.

Start, h80 [ h40 | WR ], h05, h10, h00, Stop
Start, h80 [ h40 | WR ], h00, h3C, h1F, Stop
Start, h80 [ h40 | WR ], h02, Stop
Start, h81 [ h40 | RD ], h27, hD2 NAK, Stop
Start, h80 [ h40 | WR ], h05, h10, h00, Stop
Start, h80 [ h40 | WR ], h04, Stop
Start, h81 [ h40 | RD ], h02, h9C NAK, Stop
Start, h80 [ h40 | WR ], h02, Stop
Start, h81 [ h40 | RD ], h27, hCA NAK, Stop
Start, h80 [ h40 | WR ], h05, h10, h00, Stop
Start, h80 [ h40 | WR ], h04, Stop
Start, h81 [ h40 | RD ], h02, h5D NAK, Stop
Start, h80 [ h40 | WR ], h02, Stop

bwack, 2018-11-03:
This is from with your code. I have replaced the pullups on my board now. oled disconnected.
```

Start, h80 [ h40 | WR ] NAK, h05 NAK, h10 NAK, h00 NAK, Stop
Error
Error
Start, h80 [ h40 | WR ] NAK, h00 NAK, h3C NAK, h1F NAK, Stop
Error
Error
Start, h80 [ h40 | WR ] NAK, h01 NAK, Stop
Error
Error
Start, h81 [ h40 | RD ] NAK, hFF, hFF NAK, Stop
Error
Error
Start, h80 [ h40 | WR ] NAK, h02 NAK, Stop
Error
Error
Start, h81 [ h40 | RD ] NAK, hFF, hFF NAK, Stop
Error
```

discoHR, 2018-11-03:
I tried with TinyWireM for Arduino, same problem. It's highly unlikely that both codes are faulty. The problem is most likely on the addon board itself. I've replaced 3 INAs so far and none of them sent any ACKs ever no matter what code I use. Please fix the board.

[image]

[image]

bwack, 2018-11-03:
let me try this first.

bwack, 2018-11-03:
SDA and SCL are mixed up.

bwack, 2018-11-03:
Regarding the title. The two libraries go well together. Example shown in the hans-doodle branch.
https://github.com/bwack/C64-Saver-bwack/tree/hans-doodle/addon%20board/src

bwack, 2018-11-03:
I'm closing this issue. Open a new one if you encounter a new problem.

closed 2018-11-03 by bwack