bwack/C64-Saver-bwack #21: U4,U5 EL1012 Discontinued

wishus3, 2019-05-18:
As the title states, the part as referred to in the BoM at https://www.distrelec.biz/en/optocoupler-sop-4l-80-everlight-electronics-el1012-ta-vg/p/11049956 is discontinued.
The schematic has parts U4, U5 referred to as TLP291.  Any TLP291 available at Digikey is a 4-SOIC (0.179", 4.55mm Width) that is NOT the same footprint as that on the PCB for the addon board.
Can you suggest any alternatives that fit the parameters of the EL1012 that is readily available and are the same package and size?

bwack, 2019-05-18:
The package is SO-4_7.6x3.6mm_P2.54mm. The online part selectors aren't very helpful to distinguish between 2.54mm and 1.27mm pitch and 7.6 long body. It makes it hard to find.  I see what you mean regarding TLP291. I found a LTV-817S on digikey. Same pitch, same length, might be a bit fatter. I'll try some measurements in kicad.......... 

[image]

[image]

ouff, it might be abit too fat. Let me know if you find a better fit. Electrically all of these optoisolated phototransistors will work. Timing and absolute max voltage is not a problem (slow and 5v).

bwack, 2019-05-20:
I'm changing the package to Package_SO:SOIC-4_4.55x2.6mm_P1.27mm in the next version.

closed 2019-05-20 by bwack

wishus3, 2019-05-23:
I scoured Digikey and found the following:
https://www.digikey.ca/product-detail/en/vishay-semiconductor-opto-division/TCLT1012/TCLT1012CT-ND/8556846

I used the parameters to arrive at the substitute:

    Part Status:   Active
    Supplier Device Package:   4-LSOP (2.54mm)
    Number of Channels:   1
    Current Transfer Ratio (Min):   63% @ 10mA
    Current Transfer Ratio (Max):   125% @ 10mA

I am thinking this might be an acceptable replacement.  Would this work?

bwack, 2019-05-23:
They are very close in specs. Yes use those! Thanks for searching for it.