bwack/C64-Saver-bwack #22: Removed capacistor

SFK64, 2020-04-03:
Hi bwack
The capacitor you removed to make room for resistors, what was the function of that capacitor?
Regards
sfk

bwack, 2020-04-04:
This one ?
```
C64 Saver 2.4 (2018-09-23)

*   Support lower Vth on Q1. Resistor divider R7 and R8.
*    Remove C1 to make room for resistor divider.
```

bwack, 2020-04-04:
This is the PDF of v2.3
https://github.com/bwack/C64-Saver-bwack/blob/f3cd1549b32b5a40bd19ed1b33fe1eb53d38a0f0/C64%20Saver%202/C64Saver2.pdf

I think I ment to say removed C2 not C1.
I think it was there to stabilize the gate of the NMOS transistor, but a transistor there together with a resistor divider would slow down the reaction time.

SFK64, 2020-04-27:
Yep must be C2 :)
Ok thank you very much for the answer.

bwack, 2020-04-27:
Thanks for letting me know :)

closed 2020-04-27 by bwack