bwack/C64-Saver-bwack #25: Optimizing gate voltage for Q1

franky1969, 2020-11-12:
I have built the Saver 2.4 Throughole but the output voltage was  not disconnected at overvoltage. 
I meassured the output voltage of U1 (TL431) at overvoltage with 1,8V. The gate voltage of Q1 (2N7000) was after the divider 33k/82k at about 1.3V.
According to the Datasheet of the 2N7000 the Gate-Threshold Voltage of  is between 0.8 an 3V
I did not find the the max output voltage of U1 (TL431) in the datasheet. I think this is about 2V. 
I think the values of the divider resistors should be changed. I changed the 82k to a 33k resistor this works with my parts.
Maybe further investigation is necessary.

bwack, 2020-11-14:
Thanks a lot for letting me know! I think I'll change the 2n7000 to 2n7002 due to the wide spread Vgs threshold.

franky1969, 2020-11-14:
According from the Datasheet from National Semi the 2N7000 is a pin version and the 2N7002 the SMD version.
The threshold Voltages are nearly the same:
2N7000 : 0.8V - 3V 
207002 : 1V - 2.5V
The min. Voltage maybe the reason why there are less problems witch the SMD Version.

bwack, 2020-11-14:
You are right. The problem is also with the smd version.
I think if I can't find another nmos, I can go with an npn bipolar transistor. The resistors are there anyway, so it would be a job to ensure Vth, I_b, hFE and I_KA of the tl431 (which is plenty anyway).

bwack, 2020-11-14:
V2.5 for reference:
[image]

franky1969, 2020-11-14:
The Voltage divider i have chosen has an factor of 0.5 with R7/R8. So we can get 1V this is at least on the min. threshold of the MOSFET but in worst case we need 6V (threshold max. 3V) to switch the MOSFET.
So the divider is no save solution for the MOSFET but witch an bipolar transistor it should work. 

I think, i have another possible solution for the SMD Version. In the Wikipedia artikel about the TL431 i found this:

"On-state VCA drops to around 2 V, which is compatible with Transistor–transistor logic {TTL) and CMOS logic gates with 5 V power supply.[40] Low-voltage CMOS (e.g. 3.3 V or 1.8 V logic) requires level conversion with a resistive voltage divider,[40] or replacing the TL431 with a low-voltage alternative like the **TLV431**.[41]"

bwack, 2020-11-24:
Very interesting. I looked it up. One reason for choosing TL431 is the high V_KA absolute max ~40V ish, while tlv is 10V. It would need some clamping i guess. I think maybe a bipolar is fine.

macacob, 2022-05-02:
I recognize this is an old post but I am experiencing the same issue. I tested up to 7V and output voltage was not disconnected.  I'm going to try replacing the 82k resistor with a 33k as suggested by frankly1969.  I have rechecked my work and everything seems correct (correct components, good solder joints etc).  Any other suggestions?

Thanks very much for your work on this by the way. Really a great project that should help keep more C64s alive for years to come.

bwack, 2023-03-19:
Has anyone looked at this problem? In short, the challenge is that the "output" of U1 does not go to 0 V, it goes to around 2V and varies with device to device. Then there is the threshold level of 2n7002, which have worked for me, but because of tolerances it may not work for another board. Suggestions welcome. I might have a closer look myself. 
The pandemics and chip shortages amongst other events killed this project for me. I see some people building it over at forum64, and this might kickstart some interest to continue (or not).

bwack, 2023-03-19:
I've been simulating a tl431 with cathode voltages from 1V to 2V. With a bipolar npn transistor, the threshold is a lot more stable in the bipolar transistor than the nmos transistors. Also, not much change is needed to the layout (if any? I presume some pin re-arrangement). The R7 and R8 can be set to 330k and 100k.

franky1969, 2023-03-19:
Hi bwack,
do you still have the 2V problem with die bipolar solution? Maybe this is the red LED in the V2.5 Version. If there is no load connected to output, you get about 2V from the LED current trough R9/R10.

bwack, 2023-03-19:
Hi again. This problem is with the threshold of 2n222 and variation in tl431. Regarding the led problem we can have a look at that in another issue :)

bwack, 2023-03-19:
maybe i misunderstood. im playing with lt spice now. I'd like to take out the leds until i figure out the problem with tl431 and 2n2222.

franky1969, 2023-03-19:
Sorry, i miss understood your post about the output of 2V from U1. Forget my posting about the LED it has nothing to do with this topic.

bwack, 2023-03-19:
No problem. I'm simulating with a npn transistor now and it can work. A bc547 seemed to work fine..

bwack, 2023-03-20:
This is what I've come up for the new throughhole version.

[image]

Two leds, powered by the active components (D1, Q1). Removed zener at Q1. Rotated the two TO92 packages to block less of the ground fill. Believe it or not I managed to get all components on the pcb and route them. not pretty but functional i hope.

[image]
[image]

franky1969, 2023-03-20:
Looks very nice. I have put this in my ltspice, which i found from two years ago. The BC547 is switching at about 2.5V. So this should work fine.

bwack, 2023-05-20:
I'm testing now and it seems to work, however I cannot test for parameter spread thouhg. I noticed that the V_DS was high. I'll make another issue about it.

closed 2023-07-18 by bwack