bwack/C64-Saver-bwack #26: 2.4 Throughhole C2

mikedarzen, 2021-03-25:
Thank you for creating this project.  I'm in the middle of the build of V2.4 Throughhole, and I think C2 in the BOM might be an error.   I don't see where this fits into the actual circuit/circuit board.  Is C2 needed/preferred somewhere?

mikedarzen, 2021-03-25:
Also THANK YOU for making this.   I tested my power supply a week ago and it was right around 5V.  I haven't used it since.  I tested this with a regulated power supply and it performed just as expected.  I plugged in my c64 power supply and it showed no pass through voltage on the 5V rail.
I tested my c64 power supply directly and it is now showing 12V on the 5V pins!  I'm so glad I decided to build this before I did any further testing.

MindFlareRetro, 2021-03-25:
> Thank you for creating this project. I'm in the middle of the build of V2.4 Throughhole, and I think C2 in the BOM might be an error. I don't see where this fits into the actual circuit/circuit board. Is C2 needed/preferred somewhere?

Hi Mike. If it helps, I have been using this BOM with great success, http://bit.ly/mfr-c64saver-24-bom. I also did a video series of my experience if you are intersted, https://bit.ly/mfr-series-c64saver.

Regards,
Mark / MindFlareRetro

mikedarzen, 2021-03-25:
That's awesome, thanks Mark!  I'm looking forward to watching your series.  I always enjoy watching what people are doing with their c64s.

bwack, 2021-03-25:
Glad you like it. The C2 was removed in V2.4. I have forgotten to edit it out of the BOM.

closed 2021-03-25 by mikedarzen