bwack/C64-Saver-bwack #30: High V_DS(on) voltage

bwack, 2023-05-20:
I measure V_DS=200mV at V_GS= -5V and I=1.2A.
We would like the voltage drop over the C64 Saver to be as low as possible, in case someone is using PSU with voltage on the low side of the VDD tolerance (which is 5V +-0.25 V ?)..

In Saver 1.3 I did a lot of testing for voltage drop over the whole circuit and was quite happy with it with using IRF9332. It promised Rds(on)=28.1 mOhms at V_GS=-4.5V. The C64 draws about 0.8A. For 1A draw, we would hve 30mV drop or less. Unfortunately this has a different package than what I'm using now. 

Then when I made the Saver 2 I selected a part that would give similar results. So why this high voltage drop now ?
Has the part changed ? Are there variations to the IRF5305 ?
Nobody has complained about this. I guess it works well for those who use it, but the thing is, I built myself a PSU after starting this project, and haven't been using it much myself, but when I have used it, It has been working fine.

This needs more investigation, and lets continue figure out what happened, and event better, a replacement part for this mosfet.
I would like to see a P-Channel mosfet, V_DS max 30V, R_DS(on)<=30mOhm @ V_GS=4.5V. Imax is not so important because it is usually very high for these low resistance devices.

EverlastEngineering, 2023-06-17:
Hey Bwack!

I just ordered a short run of the new 2.6 TH board with enough components to build 5 to test out the new design. Is there anything in particular I should look for?

I've made a batch of these from an order on the back in 2020 (so, version 2.4? I forget tbh) and I test each one with a small and large load. I consistently find  a sharp cut off from between 5.39 to 5.44, which drops to 100mV then 0 after a moment.

I've never seen a voltage drop from in to out less than 10mV with my builds.

I'm unsure if this applies, but I toyed around with adding LEDs myself, and found issues no matter how I routed the "red" led. I would have voltage bleeding, or affect the V drop. I don't clearly remember anymore but when these boards come in I'll be able to compare 2.4 to 2.6 and see what I see!

I might even be able to swap some components that are the same to ensure it's not a variation of the components themselves if I find anything.

EverlastEngineering, 2023-06-17:
I'll have to confirm this, but it appears I used NCE55P15K from http://www.ncepower.com/.

bwack, 2023-06-21:
Not sure of anything obvious to look out for. Hopefully the latest version 2.6 is stable now.
The only issue I've found now is what this github issue is about, the voltage drop over the saver when the C64 is on and running.
One thing is that the voltage drop produces heat. If the PSU normally just supplies 5.00V or less, the voltage seen by the C64 can be too low. 

You can use the IRFR5305 if you have a psu that produces more than 5.00 V under load. I'm using a custom PSU that produces 5.01V under load. The voltage is 4.76V when I measure over bypass caps near the chips. it works, it's not great.

I will order and try these mosfets:
https://no.mouser.com/datasheet/2/196/Infineon_IPD042P03L3_G_DS_v02_02_en-3164400.pdf
https://www.vishay.com/docs/67014/sqd50p03.pdf
https://no.mouser.com/datasheet/2/408/TJ40S04M3L_datasheet_en_20200624-1858407.pdf

LEDs:
I have also seen this issue with the bleeding in V2.5.  In V2.5 I was trying to save a bipolar transistor, but as you saw, it is impossible to turn the led fully off, the load (c64) can provide a path to gnd. That is something I've fixed in V2.6. I haven't tested the V2.6. I have tested the OCP V0.1 board, which has the same saver and LED circuits, and the leds worked fine there. No bleading there.

EverlastEngineering, 2023-06-30:
Boards are in and one was built tonight, it came together nicely.

I don't know what I was thinking earlier, because on the V2.4 boards I've been making there is about 80mV at 1A: 5.14V goes to 5.06V.

Unfortunately with the 2.6 design I'm seeing 160mV at 1A: 5.14V dropped to 4.98V 😬 The drop increases as the source V falls too:

`Volts       Drop`
`5.43V       0.13V`
`5.14V       0.16V`
`5.00V       0.24V`

Tomorrow night I'll drop on one of the mosfets from the 2.4V builds and see if that is any different.

EverlastEngineering, 2023-06-30:
Scratch pad maths says that V drop is about 160mW. Am I reading the datasheet for the IRFR5305 correctly, that the power disippation is 110...Watts?

EverlastEngineering, 2023-06-30:
Alright, enough spamming from me but after reviewing the datasheets, the 5305 specs the RDS(on) at VGS = -10V, ID = -16A.. and that's it. So I guess it shouldn't be a surprise. I think I got lucky with the NCE55P15K. I hope I see the same result from it on the 2.6 as the 2.4, then we'll wait to see what those three you order are like. I'll likely poke around LSCS and see what else is available. Maybe I'll look for something with a much lower current capability. It may yield something with a more favorable RDS(on) in the lower current and voltage range we're working in.

EverlastEngineering, 2023-07-01:
Looking through datasheets, not a bad way to spent a friday night. :) 

Question: forgive my lack of knowledge around this, but why the specification for VDS max of 30V? If 20V was tolerable I see a number of SOT-23 that have specs at 4.5V and low current. SOT-23 is a mild PITA to solder for newcomers but no worse than their bigger cousins. For me the tend to disappear when tweezers slip. 🤣 

I've noticed that with a lot of the big package MOSFETs do not have data plotted below 2A or even 5A, should they even _have_ a RDS(on) graph.

Those that do, often spec their RDS(on) resistance at least 10A, so could we expect the others without a graph to have similar specs at lower current? In other words, I draw the conclusion that many that have the high current capability to be unsuitable for lower current applications.

I'm off to build another 2.6 board, and I'll use a NCE55P15K from my last batch and then do a proper test tonight on the scope with time to voltage cut and look for any other anomolies.

EverlastEngineering, 2023-07-18:
I used the old mosfet from my last batch in a V2.6 board tonight, and I see the same 0.08V drop I did as in the v2.4 design.  I do notice a higher cutoff at 5.45V. That is the highest I've ever seen, the previous maximum was 5.44V and they're nominally 5.42V. 

This test was conducted at a 1A load, btw. I think I'll just order another batch of NCE55P15K.

Did any of the fets you order come in?

bwack, 2023-07-18:
Hi. I've been on vacation. I will order components tonight. Also I will order a new revision of the OCP version.

Regarding the 30V spec, is because of high voltage spikes when power is switched on and off.
Thanks for checking.

Wait with ordering components. The NCE55P15K is not great, because I think you mention above as well, the <75 ohm resistance is at Vgs = -10V, meaning it will get even worse at less Vgs=-5V.

EverlastEngineering, 2023-07-18:
Hope you had a great vacation!

No, the NCE55P15K is the one that I'd been using in the past, 0.065ohm at -10V/16A. It doesn't have a spec on it's datasheet at 5V / 1A but reveals ~ 0.08ohm.

The reason I asked about the 50v / 30v maxiumum is due to the greater number of mosfets with  specs in the lower V/I range that have a 30V maximum. Have you seen spikes higher than this? If so, wouldn't the C64 have seen those spikes and died? :o

bwack, 2023-07-18:
I didn't realize that you ment the power mosfet. I don't remember the exact measurements.
I've made videos about the spikes a long time ago. The spikes are seen when the power is switched on or off with the c64 power switch or the saver itself switches. The C64 does not see this spike, but the main mosfet gets it over its drain source junction. It's a kick back from the long power supply cable when the current path is disrupted. Another thing I've seen in the scope is when the power is switched on, there is a huge inrush current. This is no issue for the mosfet, but keep in mind that the power switch is not perfect, so while this inrush is ongoing, the power switch bounces for a few milliseconds. Spikes are seen at the outside of the C64.

EverlastEngineering, 2023-09-08:
Heyo! Did you get a chance to review the power loss on those other MOSFETs?

jdecupere, 2023-10-17:
Was reading this exchange with some interest as I am thinking of modifying your design to be really minimal to build into the case…so not doing the pin headers and 9v AC pass through. That would shrink it quite a bit. 
have you considered the IRF9310? I know the SO-8 is less convenient, but besides the gate pin, most other pins tue together anyway. But that one specs at below 10mOhm at Vgs of -4.5V

EverlastEngineering, 2023-10-17:
Exactly what I'm thinking of doing too, spread out the layout a bit, use wide spread legs on U1 / Q1 etc. Agree on the package of the FET too, I think the TO-252 package tends to lean towards high current applications.

That said, the IRF5305 gets mega hot at 1.5A, but that's due to the V drop.

I'm leaning towards the IPD068P03L3G, similar specs to the IRF9310.

I've just imported the layout into EasyEDA and will be working on it tonight, so I'll post here!

EverlastEngineering, 2023-10-17:
One other thing: the current going through LED2 limits the overall current capability so I'm going to move it directly to the safe 5V side of the mosfet. I prototyped it last night and it works great.

EverlastEngineering, 2023-10-18:
@jdecupere That IRF9310 looks great if you decide to do a layout with it! At the lower current needs, it looks like it would be less than 6mOhm! 

I honestly would prefer soldering a SOP than the TO-252 as it needs sooo much heat.

jhiggason, 2023-10-18:
Hey fellas- closed out my other problem as I think I have resolution on that.  Bought 5 sets of the components for version 2.6 but based on this problem would it be wise to wait for a revision or can I just build knowing that it will get hotter than it should?  It will still properly protect my c64 right?   

Based on this conversation it looks like as long as my PSU is outputting >5V @ 1amp load this still should work properly.  Is that correct?  Anything else I need to keep in mind?

jdecupere, 2023-10-18:
re-reading @bwack and @EverlastEngineering comments I am not so sure anymore it will work. It will definitely protect and not damage you C64.
But seems like the suggested IRF5305 was not tested by @bwack and @EverlastEngineering has proven it gives a large drop - even worse for lower input voltages (which makes sense as that also reduces the Vgs for the transistor and hence increases its Rds_on). So you will get issues with supplies 5V or less - and/or if you add more peripherals that draw current from this supply (e.g. on the cassette port).
I have no (short term) time to experiment, but I think @EverlastEngineering is actively trying options in the same footprint. My suggestion needs a board re-layout (for which i need to free up some time....)
If you have all the material, it won't hurt to build one and use it. if that is your only short term safe option to use an old power brick...worst case it gives your C64 not enough voltage and you will see non functionality or strange behaviour (e.g. due to timings being off)

jhiggason, 2023-10-18:
@EverlastEngineering It's totally OK, I could have waited too. =)  I've got enough parts to at least try the hack job with the right mosfet.  Would you be willing to write out/sketch what changes need to be made please?  Excited to give this all a go =)

EverlastEngineering, 2023-10-18:
@jhiggason I'll snap a picture of the LED bodge tonight for ya, it's easy!

EverlastEngineering, 2023-10-24:
@jhiggason Sorry for the delay my dude, life finds a way to get in the way. :)

[image]
[image]

There’s two cuts, one below R10 and another to the left of the big resistor. This strands the zener so I connect it to the 5v in the headers. 

I put the mosfet a little crooked so I’d have room to attach a wire to the safe 5v and ran it to the led. 

Works well and is stable no matter the value of r10.

EverlastEngineering, 2023-10-24:
Here’s a close up with the traces you need to cut circled. 

[image]
[image]

jhiggason, 2023-10-24:
@EverlastEngineering thank you so much! This really means a lot to me.  Hopefully I'll be able to post about success shortly!

Jeff

PS do you feel like we can order the 2.7 board yet?  I'd be willing to give it a go if you think it's good.

EverlastEngineering, 2023-10-24:
Wait a couple days and my prototype boards will arrive and then we'll know for sure! :)

jdecupere, 2023-10-27:
HI there
I am waiting for the boards to test the IRF9310, but meanwhile breadboarded the rest (so, all is included except the IRF9310) just to do some checks.
Just thought I'd share the quick check on startup behaviour here with some screenshots.
I used @EverlastEngineering suggested resistor split and 330 ohm +3.3k. This seems to give a trigger point around 5.28V in my breadboard.
Seems to work nicely and quickly :). In the final config, there will be a big cap from the C64 on the safe 5V as well as all kinds of load....so there would be no risk at all of getting even transient overvoltages it seems.

Blue CH2= 5V "unsafe"
Yellow CH1 = gate of the MOS power FET

V_in to 5.2V:
[image]

V_in to 5.35V:
[image]

EverlastEngineering, 2023-10-27:
^^ Awesome!

My boards arrived just now, I'll be building one tonight with a leftover mosfet that has the higher V drop to test out the rest of the changes. 

In regards to scoping the results, I found something really interesting likely due to the cap on board buffering the voltage. When I first tested the 2.4 design I saw a spike, sometimes _considerable_ in the `5V_Safe` but I had no load at all while testing; that seems to be critical. With any appreciable load, right up to 1.2A, there was no spike at all. I'll see if I can get get a quick video of it.

@bwack why don't you open `Discussions` on this repo so we can carry on the convo's that aren't strictly related over there?

bwack, 2023-10-27:
Done, added discussions.

bwack, 2023-10-28:
No worries. You can discuss here if you like.
I have populated my c64 saver OCP v0.2 board. If I can get it up and running I will post some results.

bwack, 2023-10-28:
Having fun testing now. Just compared the IRF5305 to TJ40S04M3L. Pulling about an amp (0.99A). Seeing V_DS go from 218mV to 11mV !! Respectively! Wow!

EverlastEngineering, 2023-10-28:
We should make a wiki about mosfets that are known good and what spec to
look for when searching for alternatives. It’s a great way to learn about
data sheets.

bwack, 2023-10-28:
I invited you as collaborator. If you like to edit the wiki.
I'll put the alternatives i find into the boms.

Another measurement, same conditions as above (0.99A, VIN=5.01, V_GS=4.93):
SQD50P03: V_DS = 8 mV

bwack, 2023-10-29:
IPD042P03L3GATMA1: V_DS=4mV !

EverlastEngineering, 2023-11-05:
I've added a wiki with these values, so I'll close this issue now!

https://github.com/bwack/C64-Saver-bwack/wiki

closed 2023-11-05 by EverlastEngineering