bwack/C64-Saver-bwack #31: BOM?

jhiggason, 2023-10-16:
Hello!  First of all thank you for providing this.  It's a really great piece of equipment and I'm excited to try it out on my "new to me" c64 =)  

I'm a very experienced electronic hobbyist- but building a board with Gerber files and parts is really new to me.  Do you have an actual part list somewhere?  Because I see something called a kessler number which doesn't seem to match up with any of my vendors for parts.  Let's say for C1.  It shows it is a 47UF cap that is CP_Radial_D6.3mm_P2.50mm....am I just going to mouser and trying to find caps that are round and the right size for the uF?   There are so many millions of parts that the options are a bit confusing for me.  

Any help would be great appreciated. Thank you =)

jhiggason, 2023-10-16:
Ok folks- for others that might be someday joining in- I'm learning some specific things about components with gpt4's help.  For instance:  C1 would be:

47uF: This is the capacitance value of the capacitor.

CP_Radial_D6.3mm_P2.50mm: This suggests a few things:

Radial: This is a type of capacitor where the leads come out from the same side (as opposed to axial capacitors where the leads come out from opposite ends). Radial capacitors are commonly used in printed circuit board applications.

D6.3mm: This indicates the diameter of the capacitor is 6.3mm.

P2.50mm: This indicates the lead (pin) spacing is 2.50mm apart.

Then I could into mouser with this new information and start setting filters.  It doesn't say this explicitly but a decent cap would probably be at least 1.5 to 2 times the voltage requirements. Since we are at 5v, a 10v cap or 16v cap would work great.  25v is still fine and might be cheaper than some others.  (You can shift click mouser filters and select all 10v, 16v, and 25v caps)  

The only other important filters are hours between failure (typically 1000 to 5000, the later being more expensive)  ESR and Ripple current measurements.  I'm not sure how important these are for this exact application but gpt4 thought a lower ESR and high ripple current is safer and would typically cost more.  Also temperature ratings as well.  Pick whatever.

jhiggason, 2023-10-16:
Here is a cap for C1 I found at mouser: https://www.mouser.com/ProductDetail/710-860240473003

jhiggason, 2023-10-17:
Ok I found on the SMA BOM it has an alt for the ATPAK-2 - so it must not be required.  Attached here is my BOM- I didn't include the 4pin and 8pin headers as I already have those.  If someone would be kind enough to cross reference my parts I would be eternally grateful.  Otherwise I'll probably just order and give it a go.  

This is for Mouser.com

[Mouser_Cart_2023.xls](https://github.com/bwack/C64-Saver-bwack/files/12938158/Mouser_Cart_2023.xls)

jdecupere, 2023-10-17:
You should be fine. Atpak2 is the same footprint, just lower in height, I think.

EverlastEngineering, 2023-10-18:
Heyo!! Hopefully this catches you before you order. We've found the IRFR5305 isn't suitable, if you want the details you should check out the other open issue here.

Also, I found the cap to be a big too tall to fit in the case which is the standard hammond case size so I'll be looking for a slightly shortly one on my next order.

Finally, I found a couple small issues with the 2.6 design and have spent the night redesigning it a bit. So, perhaps hang on just a bit until this gets sorted? I'm about to place an order in the next day or two for pcbs and parts and then I'll know more.

Here's the redesign from tonight's marathon lol:

[image]

jhiggason, 2023-10-18:
@EverlastEngineering Oh noo!  I just ordered all the parts and the board, figured I'd give it a try.  I ordered enough for 5 boards...   Do you think I can use these parts on the new version or somehow make the version I have work?  I can also just order the new part and put in another order for the new version of the board if you think that would be better? 

You are a rockstar btw!  Thanks again for all this.  I plan on writing some documentation for the through hole build I'm doing.  Are you guys in the US?

jdecupere, 2023-10-18:
I am no GitHub expert, but maybe it is better to continue discussing in the other issue (30) as what we discuss relates to the findings there. 
meanwhile answering your last questions: it will work, but has some shortcomings (see issue 30): due to the ON resistance of the transistor you may have issues with power supplies that are on the lower side of the 5V spec…and the C64 Saver will get hot. @EverlastEngineering is also doing some other improvements.
(I am in Belgium btw)

jhiggason, 2023-10-18:
Yeah I think we can can close this issue and if I have any questions relating to issue 30 i'll ask there.  I'm new here- don't mind me ;)

closed 2023-10-18 by jhiggason