bwack/C64-Saver-bwack #33: R10 in C64 Saver SMD 2.5

jueank, 2023-10-27:
Hi,
I'm referring to the 2.5 SMD version of the saver. Why is R10 (3k3) directly connected to 5V_SAFE, while on the PCB it's routed to positive pin of LED2 but merged with the 5V_SAVE plane? Negative pin of LED2 is connected through R11 (3k3) to GND. Is this an issue? If not, can you please explain?

jdecupere, 2023-10-27:
Hi,
As I was looking at the design anyway, I did a quick check - for the reasoning behind the design choice @bwack will need to answer.
I am not sure I understand the first part of your question. There certainly is no issue. The PCB connections are the same as the schematic. R10 connects to 5V_safe, as well as the LED2 anode...on the PCB one is drawn as a wire, the other through the fill plane...but in the end those are merged as 1 Cu layer (you can check the gerber file if you want to see how it looks), so it does not matter how the PCB trace is drawn.
Schematic-wise, having a resistor between 5v_safe and GND seems to be done to provide a path for the current to go to in case the input voltage is above the safe trigger point:
- If V_in > trigger, Q2 will be OFF, and the current will flow through LED1, R9 and R10 to GND -> LED1 should light up to indicate the "fault" situation. 5V_safe will be at some intermediate voltage - a little lower than V_in/2 (without additional load) but with high impedance so in case you have the C64 connected, any current draw will lower the voltage even more --> hence all is safe
- if V_in <trigger, Q2 will be ON and hence LED1 and R9 will not see a voltage across them and the LED is OFF; V_in is passed to 5V_safe and there will be a small 3mA current through R10 as "waste" and another path through LED2 and R11 to show that the 5V_safe is "live"

I think the later designs are more elegant, but this should not have an issue per se

BR,
Joost

jueank, 2023-10-27:
Thanks for the explanation. I know, the trace frm R10 to LED2 is merged with the 5V plane. I was just confused by the way it was done. Not sure if bwack has to add something to what you wrote!

BR :)

hansdals, 2023-10-27:
Not sure without digging up 2.5 from 2020. It could have been I had limited routing space. Also the led circuit was a late fix, snd wasn't done very well in 2.5 unlike 2.6.

bwack, 2023-10-27:
Not sure without digging up 2.5 from 2020. It could have been I had limited routing space. Also the led circuit was a late fix, and wasn't done very well in 2.5 unlike 2.6.

bwack, 2023-10-27:
Just to be clear, are you saying that the schematics and layout are different?

jueank, 2023-10-27:
No, there is no difference. It is just that the direct trace from R10 to LED2 - although merged with 5V plane - is explicitely routed, indicating a relationship of LED2 to R10. That COULD have been a mistake. I just wanted to be sure and asked therefore here.

bwack, 2023-10-27:
Oh as in the way the schematic is read it can look like there is a relationship, not intentionally. R10 was put there to provide a path for LED1 to GND even if the load is disconnected.

jueank, 2023-10-28:
Thanks for all so far. Since there is no fault, I think the issue can be closed. @bwack can I ask you something to the schematic then in a private discussion? I'll write it in a minute...

closed 2023-10-28 by jueank