bwack/C64-Saver-bwack #34: Through hole Zener Diode Sizing

jhiggason, 2023-10-27:
Hey fellas.  Just getting around to putting this thing together.  The BOM for through hole zenners isn't very clear so I went to SMD bom and found the part number that was provided there.  Then I bought a zener based on that.  I only hesitate because this one is much smaller in size than the markings on the board.  Both are showing .5watts 6.8v 2% so hoping I'm OK.  

What do you think?  

The part I bought: https://www.mouser.com/ProductDetail/Vishay-Semiconductors/BZX55B6V8-TR?qs=WLTup5mM6hj4D2ZUwZwaKA%3D%3D

The part I referenced from the SMD bom:  https://www.mouser.com/ProductDetail/Taiwan-Semiconductor/BZY55B6V8-RBG?qs=mAH9sUMRCtuk9q3vqhCMbg%3D%3D

Thank you!!

EverlastEngineering, 2023-11-05:
Mmwm.... the part you bought is 1/2 watt while the one referenced in the kessler column of the bom is a _3w_. 

I got the three watt one, but dannnnng he a big boy. 🤣 I'm thinking that it might a mistake, the V2.4 of the board spec'd two smaller ones and if the SMD also specs a small, I'm sure I'll be fine.

jhiggason, 2023-11-05:
So I have looked in detail at what the heck a Kessler number is...it doesn't seem to have a meaning in the US.   What website did you use to find a part based on that?  I'm thinking it might be an overseas vendor?

Thanks for your help on this.  I'll try out the .5w

jhiggason, 2023-11-05:
I found a reference to Kessler in your new wiki post. Others can go there to found out more.  Thanks!  https://github.com/bwack/C64-Saver-bwack/wiki

jhiggason, 2023-11-06:
@EverlastEngineering So thank you for providing that the kessler number was a german company + the link.  That has helped immensely.  For some reason, even knowing that it was a german company, I could find no record of it on google and even chatgpt was stumped.  It was driving me a bit bonkers.  

So I guess a few things- the kessler number for zener diode is actually a 1.3watt device.  DO41 https://www.kessler-electronic.de/de/Bauelemente/Halbleiter-diskret/ZPY-6-8V  

This is showing something different than even the bom which is showing a DO35 zenner.  I couldn't find 1.3 watt zeners that ship quickly so I went with a 1watt version of the same form factor (DO41).  Hoping that lines up better with what was planned. 

@bwack How important is the 1.3watt zener for the safety of the circuit?  Will my 1watt be fine?

jhiggason, 2023-11-06:
Ok a bit more information- there appears to be 2 part numbers for the SMD version zener. 300mw 6.8v SMD https://www.digikey.com/en/products/detail/nexperia-usa-inc/BZX384-B6V8-115/1162926  

Here is the other it references:  500mw 6.8v 2% BZY55B6V8  https://www.digikey.com/en/products/detail/taiwan-semiconductor-corporation/BZY55B6V8/18720320

So I'm not exactly certain.  You guys have any thoughts on this?

jdecupere, 2023-11-07:
Hi Jeff,
(note: this issue is closed, so maybe better to discuss in the discussions section)
As the zener is just meant to handle spikes (pos and neg) and will never have to pass large current, I don't think the Wattage rating matters too much. You want to have 6.8V breakdown to avoid issues in normal operation and other than that just the effect of (somewhat) reducing influence of the potential spikes - as bwack mentioned this can happen during hot swapping the connector.
I have even ordered a 0.5W one myself - boards still have not arrived, so cannot comment on actual measurements yet.
(mine is this one: https://www.reichelt.com/be/en/zener-diode-0-5-w-6-8-v-zf-6-8-p23142.html?r=1)

jhiggason, 2023-11-07:
@jdecupere Thank you for responding. =)  I'm definitely not an electrical engineer...so just making sure I'm not going to damage my C64 baby =P   I wasn't even sure how to test that the zenner was high enough wattage so figured it was wise to ask.  I will put together a wiki article on everything I've learn for other newbies like me once I'm done here. Thanks again.

jdecupere, 2023-11-08:
@jhiggason no problem..ask away! :)
almost 30 years experience in semiconductor (analog) design and electronics in general on my side...which does not mean I know everything, but glad to (try to) help anytime

bwack, 2023-11-08:
Maybe there are some zeners that are more suited for high voltage transients? Like tvs? I'm not too familiar with them. But yes th 3W i think is not important

jhiggason, 2023-11-13:
Thanks- I'll open a new request if I have any questions  Going to test with the 1watt.

closed 2023-11-13 by jhiggason