bwack/C64-Saver-bwack #36: Soldered board (2.6 throughhole) with new mosfet not working =/

jhiggason, 2023-11-13:
Hey Fellas- I got my board together and tried a test for the first time last night.  I'm not getting any indication of power on the board.
I made the two cuts, installed a IPD068P, changed the zener location, and hooked up the LED but nothing.  I had the hardest time getting the two small transistors soldered(Q1, U1)  I'm wondering if I possibly heat damaged them?  

Any troubleshooting steps I can follow?  

I had the input hooked my bench power supply at 5V- I had a 10Watt, 5 ohm resistor on the output for a load.  Is the a proper method to test it?  If no- how would I better test the unit?  

[image]
[image]
[image]

jhiggason, 2023-11-13:
@EverlastEngineering Did that new pcb design workout?  I might just go that route if yes...

jdecupere, 2023-11-14:
@jhiggason : this is an easy one: you forgot to solder the by-pass on the two pin header connectors that bwack has foreseen for the add-on boards to measure current. Since you do not have that add-on board, you need rot short it yourself. So on the 2 8-pin rows (left and right in your picture), short the middle two (so pins 4 and 5 if you will) - just a small wire, soldered in those 2 will do.
Don't worry about the heat damage....looks ok on the picture

jdecupere, 2023-11-14:
I was about to point you to it 😀 and then saw your edit.
I actually just remembered from the YT video that brought me to this design from MindFlareRetro

jhiggason, 2023-11-15:
Ok I have power!   But now I'm getting weird behavior-  Is there supposed to be a power led?  If yes- only 1 led appears to be working.  

Applying 5V of power- with my 5ohm resistor I'm seeing about .80amps being drawn from my psu.  Output voltage is only showing 4.8V.  If I increase my input voltage to about 5.2- I see 5V output.   Is this because of my resistor?  

When I increase input voltage- I see output voltage raise along side of it until I hit 5.6V - then the red LED comes on but nothing else seems to happen.  Isn't this the point where the circuit is supposed to cut power to the output?  Or is the purpose of the circuit to reduce power to the safe 5V?  Either way it doesn't seem to be working. Getting closer!

Thanks again for the help fellas.

jdecupere, 2023-11-15:
If I remember correctly from the other discussion on the BOM, you use the IRFR5305, correct? This is what @EverlastEngineering found out already: it has a high ON resistance with the -5V Vgs...around 0.2 ohm. This correlates to what you see as delta between input and output with your 5ohm load: drawing (a little less than) 1A gives you 0.2V drop.

One LED on when in a safe input range is normal. Should be the green one. When the voltage protection is active (so your input is above the threshold) then the red LED should be on....I think this seems to be working ok per your description.

The threshold of 5.6V seems correct as well with the BOM you used.
Are you sure nothing else happens? When you measure 5v_safe, it should now drop to 0v to close to it. Did you measure across your 5 ohm resistor?

jdecupere, 2023-11-15:
Ah...wait...you made the bodge @EverlastEngineering proposed...maybe that was not done correctly?

EverlastEngineering, 2023-11-16:
HI!! Sorry, I've been nutso busy.

Yes, the V2.7 boards seem to be working well! You can find the info in the PR that's open, but be warned, I ordered mine through EasyEDA in their own editor so it's not the _exact_ kicad file you see, I recreated it for this repo.

As far as the V drop goes, where are you measuring it from exactly? My power supply, set to, say, 5.2V will show as low as 5.0 at the input section of the board. But the real critical part is measuring the voltage drop from the 5V in to the 5V out. This is where the IRFR5305 failed badly but the ones in the wiki all seem good.  I can't tell in the pic which MOSFET you have installed.

Next, I will go back and look at your LEDs issue.

EverlastEngineering, 2023-11-16:
@jdecupere was correct, I have a jumper on the outside pins of j5, pic incoming shortly!

EverlastEngineering, 2023-11-16:
One more note: I found that the 3.3k plus the 330 ohm resistor combo is too low, 3.9 is too high so I’ve settled on the 3.3 plus a 510 which makes a cutoff right at 5.4V.

EverlastEngineering, 2023-11-16:
I just reread everything to get a clear picture. When you hit around 5.5 the red led should light and output V should go to 0. 

Question that might be super silly: did you have it hooked up backwards. I’ve done it and that’s what the effect is. It doesn’t hurt anything but I was mega baffled for a bit lol.

jhiggason, 2023-11-16:
LOL- ok you all gave me a ton to go off of.  When I get home tonight let me see if I can figure out what i have.   I did install the newer recommended mosfet.  The IPD068P.  

It's funny- I saw all those jumpers you had in the orginal pictures for the bodge- but honestly assumed you were using a different version of the board because it wasn't mentioned.  Probably should have just asked LOL.   I'm learning so much in the process though and that is exactly what this project was meant to do!  So thanks again for all the help here - you all are seriously rockstars.

Btw- I totally could have jacked up the bodge lol.  I did order the new set of 2.7 boards but it'll be a few weeks before they get here.    

@EverlastEngineering  I don't think I had it backwards because when i had issues I kind of doublechecked everything but I'll hook it up today again and try to get some clearer photos for you all so we can be sure.

EverlastEngineering, 2023-11-19:
@jhiggason Which files did you use to order the 2.7? I _HOPE_ the gerber from the easyeda folder (as that's the one I ordered.) I did a cleanup tonight and was confused by a small unattached NET in the Kicad file.. it was a wire that was over the R3 connector in the schematic _but not connected_. I really had to fight to get it to "connect" and that of course meant there was a missing trace from R2 to R3. 

The Kicad DRC _passed_ because those weren't connected in the schematic. Sheesh.. 

If you _did_ actually use the Kicad file and end up with boards with that defect, I can show you what to easily bodge, or you can use a single resistor without any bodge at all by running it across R2 and R3, skipping the mis-connected ones.

Sorry, mate, if that miss ended up costing you boards. :(

jhiggason, 2023-11-21:
I think I might have ordered the one in root of the folder.  So maybe the bad one?  It's not the end of the world..I had a coupon so it was like 3.50 shipped.  It's really no issue.  

If I wanted to skip that resistor, I can just order a 3.83k ohm?  

I haven't had a chance to look at my bodge board yet...I just got my SD2IEC setup and working and was able to play pacman for the first time last night.  Big win that has been a long time coming.  Now I just need to get this working and i'll be golden.

jhiggason, 2023-11-22:
Ok fellas- I spent a bunch of time yesterday really looking at the board and trying to figure out what I've done wrong.  I checked continuity at a ton of parts to see if maybe my solder joints were bad or to see if maybe my cuts weren't deep enough. Everything tested good.  The two cuts and all of the important parts all have continuity where they should, and no continuity where they shouldn't.    I tried to take some better photos of my board but they still are not great...I really need to get my macro photography setup going. 

Here is what is happening.  

Apply 5V to the board with a 5OHM resistor on the load.    As I increase voltage- from 5V it increases across the load until about 5.6v where the red light comes on but the voltage doesn't drop.  (green light works now btw after that jumper to ground)  Voltage will continue to increase on the output indefinitely.  

[image]
[image]
[image]
[image]

What parts do you suppose I should be looking at?  I'm guessing mosfet and zener?  Both are different than what is in the bom but I thought would work based on part numbers and other specs.   

Here is the zener kit I ended up with: https://www.amazon.com/dp/B07BTKVRG8?psc=1&ref=ppx_yo2ov_dt_b_product_details

I used the 6.8v 1watt zener.  

Here is the mosfet I purchased: https://www.ebay.com/itm/132325095444

Really enjoying this project- thanks again fellas. =)

jdecupere, 2023-11-22:
Hi Jeff,
Since the red led lights up when you go above the trigger voltage, the 431 (u1) and related circuits seem to work as expected. For some reason the MOSfet gate seems to stay low so your safe 5V is not cut off. The zener should not be the reason....even if you had it reversed or so, the result would be different (never turning the FET on fully).
What I suspect is something related to the npn(q1). Seems like the collector is stuck low. Possibly a short there? It is partially under your bodge wire in the picture so hard to tell.
Can you check the voltages on the zener pins when having >5.6V vin and hence red led ON? If one side remains low, we need to start there and see why that is the case, even with "ngate" clearly low (deduced from the fact that your red led goes on)
Br,
Joost

jhiggason, 2023-11-22:
@jdecupere  Thanks I'll look at q1 tonight.  Because that was the closest device to the bodge- I spent a bunch of time last night checking for shorts around it and seeing where voltage was flowing to and from that device to see if I goofed anything.  

I'll check across the zener with the red light on.  

One more question for my knowledge.  When I take voltage readings across the resistor- I'm getting voltage drop- which I would expect for a resistor.  But when I take the readings at the terminal block- I don't get voltage drop.  Why is this?  Is something wrong with my terminal clamps where I'm getting extra resistance?  I was taking the measurement from the top of the terminals screws.  But because there is just a wire and a resistor on the other end...shoudn't the voltage be the same??  I'm a bit confused by this.

jhiggason, 2023-11-22:
@jdecupere @EverlastEngineering @bwack I don't know how I can help you guys- but you all have been invaluable on learning more from the project and helping hone my small electronic skills.  

I have a ton of knowledge around sUAS, 3D printing, and my day job is an IT manager for a medium business.  So anything PC related/server/networking enterprise level stuff I'm a pro at.  Please reach out if there anything I can do to help you guys out.  I just really appreciate this and I know this is all unpaid time you are all offering.  It just really means a lot to me and I'd like to share the love if I can.

jdecupere, 2023-11-22:
No problem...it is actually my first time actively participating and it is fun to help :)
I am not sure I understand your question on the measurement points...do you mean you see a different reading when you measure with the probe points on the terminal screws vs measuring on the resistor terminals? With resistor terminals you mean the big one that you use as load on the breadboard, right? These should indeed be the same and probing the screws is a valid way to do it. So, unless I misunderstand what you mean, if you see a difference, then that is very strange.

By the way: I made a simplified version - I may start a discussion to see if there is interest adding it to the repo and if so I will do it formally - and works fine so far. I have not yet tested with a load (I need to get me one of the high power resistors like you). I removed all the pin headers, removed the 9V AC bypass and switched to a different MOS in an SO-8. Also removed the green led. My intention is to build these in in the C64s I saved from chip harvesters and/or e-waste disposal before I sell them (so unsuspecting users don't kill them with an old damaged supply). Here is a picture (ignore the GND and 5V pin headers - these are just for testing now...and I clearly still need to practice soldering the SO-8 ;-))

[image]

jhiggason, 2023-11-22:
Yes exactly!!  Measuring the voltage across the terminal screws vs on the breadboard with the resistor shows different voltages. When literally there is only a wire between them.  I'm thinking either a breadboard or a terminal problem..but just wanted to make sure I wasn't misunderstanding how voltage works haha.  

[image]

Oooooo- I like your new design.  If you get to the point where you want other testers- let me know- I'd be happy to order and try it out for you.  Are you in the US?  I have a pack of those resistors- I can send you one through USPS.

jdecupere, 2023-11-22:
Ok. Clear picture. It is a bit higher than I would expect, but not impossible. The green wires and the breadboard traces will add some resistance...0.4 ohm it seems. Not impossible if the breadboard is not the highest quality....or there could be some contact resistance if the green wires are not pushed in the board all the way. In any case...this is not part of the problems you see ;-)
Thanks for the offer on the resistors, but I am in Belgium, so that would not be cost effective...I need to add it to my next order from my local supplier.

jhiggason, 2023-11-23:
Ok- voltage across the zener increases until about 5.4v and then drops to 32mv.  

Output continues to increase as input is increased.

jdecupere, 2023-11-23:
Aha! I think I found it...we need @EverlastEngineering to check, since I think it is his bodge ;-)
Unless I miss something, the two PCB cuts he proposed disconnect R10 from "overvoltage" and the cathode of the zener  from VCC1. The jumper short (pin 2 to pin4) then connect R10 to GND instead -> this is OK.
To reconnect the zener, you move its cathode directly to VCC1 (as you did on the bottom of the PCB) - again fine.
Then, he says to connect the bodge wire from LED2's anode to 5V_safe, so it is powered from there, resulting in the green led being on when the voltage is safe (and in the process not pushing current into the "overvoltage" node. BUT, here is where either he missed an instruction or missed that this does not work: The anode of LED2 is still also connected to VCC1 through a trace that goes to LED1 and the de-coupling cap - i tried to indicate in green below what I mean.

[image]

So, @EverlastEngineering : can you check if there is another cut @jhiggason needed to make, or you missed this one?

Anyway: two options to move on  (choose one, not both):
1. remove the red bodge wire: this will remove the short and will keep the green LED on VCC1 - hence always ON (and not indicating safe 5V, but just that there is power on the input) - everything else will work
2. make another cut to remove the short and have the components where @EverlastEngineering intended them: cutting the trace between LED2 and LED1 that connects their pin2 (anodes) - see below

[image]

@EverlastEngineering can you confirm?

jhiggason, 2023-11-23:
This did it!  Removing the red bodge wire allows the circuit to open properly. Around 5.48V it cuts out and drops to 28mv.  It also fixed the voltage problem.  I'm now reading the same voltage at the load as the input voltage. =)

jhiggason, 2023-11-24:
Hey fellas- I was able to model and quickly print a case for it last night.  =) Does version 2.7 have the same form factor?  If yes- I can share my little project box.  Not the tightest but works great.  Thanks again for all the help.  

[image]
[image]
[image]

jhiggason, 2023-11-24:
I uploaded the STL for the case here: https://www.thingiverse.com/thing:6332407  You can either link to my thingiverse project or feel free to download the STL and place it somewhere here as well.  Glad to contribute to a great project =)