bwack/C64-Saver-bwack #38: 330K Resistor

brendan1495, 2024-04-26:
Hi, not really an issue but I am in the UK and really strugging to get hold of a 04W 330K resistor, I can get hold of 300K 0.4W will this be Ok. Sorry for the noob question.

Brendan

jhiggason, 2024-04-26:
Can you find higher wattage?  I'd have to look at the schemetic again to see what the 330k are doing...but the easiest would be looking for higher wattage resistors. 

I'm sure the other guys here can give you better information ❤️

brendan1495, 2024-04-27:
Hi jhiggason, thanks for the quick reply,  I could get higher but how high can I go?

jdecupere, 2024-04-27:
Hi Brendan,
That resistor is not critical. It just provides some hysteresis to the switching point to avoid an unstable state when just triggering the safe mode. Whether that is 330k or 300k Ohm does not matter much. Just take something close that has the same size - wattage is also not important...if you think about it - even if you had the full VCC over it, it will still just be  in the order of 0.1mW range.
Joost

brendan1495, 2024-04-27:
> Hi Brendan, That resistor is not critical. It just provides some hysteresis to the switching point to avoid an unstable state when just triggering the safe mode. Whether that is 330k or 300k Ohm does not matter much. Just take something close that has the same size physically - wattage is also not important...if you think about it - even if you had the full VCC over it, it will still just be in the order of 0.1mW range. Joost

Thanks a lot, managed to get some 300k in 0.4W. 0.4W resisters seem to be really hard to get in the UK in small numbers. Obviously I could use digikey or mouser but postage is very expensive unless buying lots. Thanks for the reply, I can get to building them now.

Brendan